dotintent/FlutterBleLib #486: 'NSInternalInconsistencyException', reason: 'Duplicate plugin key: FlutterBleLibPlugin'

gowtham2809, 2020-06-30:
Hey, I am getting the following error when I run the app on iOS. Can someone help me in figuring it out?

```
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Duplicate plugin key: FlutterBleLibPlugin'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff23e3cf0e __exceptionPreprocess + 350
	1   libobjc.A.dylib                     0x00007fff50ba89b2 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff23e3cc88 +[NSException raise:format:arguments:] + 88
	3   Foundation                          0x00007fff258b8bd2 -[NSAssertionHandler handleFailureInMethod:object:file:lineNumber:description:] + 191
	4   Flutter                             0x0000000105fa63a8 -[FlutterEngine registrarForPlugin:] + 320
	5   Runner                              0x0000000104a8d649 +[GeneratedPluginRegistrant registerWithRegistry:] + 617
	6   Runner                              0x0000000104a8e4fc $s6Runner11AppDelegateC11application_29didFinishLaunchingWithOptionsSbSo13UIApplicationC_SDySo0j6LaunchI3KeyaypGSgtF + 220
	7   Runner             <…>
```

AppDelegate

```
import UIKit
import Flutter
import GoogleMaps
import flutter_downloader

@UIApplicationMain
@objc class AppDelegate: FlutterAppDelegate {
  private var methodChannel: FlutterMethodChannel?

  override func application(
    _ application: UIApplication,
    didFinishLaunchingWithOptions launchOptions: [UIApplication.LaunchOptionsKey: Any]?
  ) -> Bool {
    GeneratedPluginRegistrant.register(with: self)

    let controller = window.rootViewController as! FlutterViewController
    methodChannel = FlutterMethodChannel(name: "https.******.tartlabs.com/cnannel", binaryMessenger: controller as! FlutterBinaryMessenger)

    methodChannel?.setMethodCallHandler({ (call: FlutterMethodCall, result: FlutterResult) in
      guard call.method == "initialLink" else {
        result(FlutterMethodNotImplemented)
        return
      }
    })

    GMSServices.provideAPIKey("**********************************")
    FlutterDownloaderPlugin.setPluginRegistrantCallback(registerPlugins)

    return super.application(application, didFinishLaunchingWithOptions: launchOptions)
  }

}

private func registerPlugins(registry: FlutterPluginRegistry) {
    if (!registry.hasPlugin("FlutterDownloaderPlugin")) {
       FlutterDownloaderPlugin.register(with: registry.registrar(forPlugin: "FlutterDownloaderPlugin"))
    }
}

```

mikolak, 2020-07-01:
Oh crap, my grave mistake. Releasing patch in a second.

mikolak, 2020-07-01:
I've fixed the issue in BLEmulator, just bump it to 1.1.3.

closed 2020-07-01 by mikolak

gowtham2809, 2020-07-01:
Hey, @mikolak I have downgraded the plugin and still, I am getting the same issue.

Pubs for ur reference
```
  # Library to support BLE communication
  flutter_ble_lib: ^2.2.5
  # Plugin for simulating Bluetooth Low Energy peripherals.
  blemulator: ^1.1.2
  # A Flutter plugin for opening iOS and Android phone settings from an app.
  app_settings: ^4.0.1+1
```

Once changed I did the following steps to make sure
```
cd ios
rm -rf Pods Podfile.lock
pod deintegrate
pod update
cd ..
flutter run
```
But Still I am getting the same error

gowtham2809, 2020-07-01:
Hey, @mikolak I have downgraded the plugin and still, I am getting the same issue.

Pubs for ur reference
```
  # Library to support BLE communication
  flutter_ble_lib: ^2.2.5
  # Plugin for simulating Bluetooth Low Energy peripherals.
  blemulator: ^1.1.3
  # A Flutter plugin for opening iOS and Android phone settings from an app.
  app_settings: ^4.0.1+1
```

Once changed I did the following steps to make sure
```
cd ios
rm -rf Pods Podfile.lock
pod deintegrate
pod update
cd ..
flutter run
```
But still, I am getting the same error

mikolak, 2020-07-01:
Run pub get and try to launch from Flutter IDE (ie. not XCode). Your `ios/Classes/GeneratedPluginRegistrant.m` should contain BlemulatorPlugin.