dotintent/FlutterBleLib #489: [BUG][IOS] Write not work

maine98, 2020-07-02:
## PROBLEM
Hi, I installed the plugin but on ios doesn't want to know how to work writing on ble, I can read but not write.
On android instead everything works properly.
I use a ESP32-Dev-Module with ble.

## CODES
```dart
//First of all, after connecting to the device, 
//I read the writing and notification characteristics 
//and save them in the appropriate variables

void readCharacteristic(Service service) async {
  await service.characteristics().then((List<Characteristic> res) {
    res.forEach((item) {
      if (item.isWritableWithResponse) {
        singleService = service;
        characteristicWritable = item;
      }
      if (item.isNotifiable) {
        characteristicNotifiable = item;
      }
    });
  });
}

//With this i read the parameter from Esp32-Dev-Module
_read() async {
    device.monitorCharacteristic(singleService.uuid, characteristicNotifiable.uuid).listen((res) {
      setState(() {
        print(utf8.decode(res.value.toList()).toString());
        value = utf8.decode(res.value.toList()).toString();
      });
    });
}

//With this i send A or B for turn on or turn off led
write(int val) async {
    return await characteristicWritable.write(Uint8List.fromList([val]), false);
}
```

## CONSOLE LOG

```dart
[CoreBluetooth] API MISUSE: <CBCentralManager: 0x2827fae80> has no restore identifier but the delegate implements the centralManager:willRestoreState: method. Restoring will not be supported
[CoreBluetooth] WARNING: Characteristic <CBCharacteristic: 0x281642580, UUID = 6E400002-B5A3-F393-E0A9-E50E24DCCA9E, properties = 0x8, value = (null), notifying = NO> does not specify the “Write Without Response” property - ignoring response-less write
```

## FLUTTER DOCTOR [MAC]
```dart
[✓] Flutter (Channel stable, v1.17.5, on Mac OS X 10.15.5 19F101, locale it)
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 11.5)
[✓] Android Studio (version 4.0)
[✓] VS Code (version 1.46.1)
[✓] Connected device (1 available)
• No issues found!
```

## FLUTTER DOCTOR [WINDOWS]
```dart
[√] Flutter (Channel stable, v1.17.5, on Microsoft Windows [Versione 10.0.19041.329], locale it-IT)

[√] Android toolchain - develop for Android devices (Android SDK version 30.0.0)
[√] Android Studio (version 4.0)
[√] VS Code (version 1.46.1)
[√] Connected device (1 available)

• No issues found!

```

## DEVICES TESTED
- Oneplus 5T (ANDROID 10) -> Work
- Iphone 6 (12.4.6) -> Not Work
- Iphone XS (13.5.1) -> Not Work

mikolak, 2020-07-02:
Hi!

You're checking if it's `writeableWithResponse`:
```
    if (item.isWritableWithResponse) {
        singleService = service;
        characteristicWritable = item;
      }
```

but writing without one:
```
return await characteristicWritable.write(Uint8List.fromList([val]), false);
```
Flip the boolean at the end of the line to `true`.

maine98, 2020-07-03:
Hi @mikolak , thank you so much for your quick response, I tested and it works correctly 💪🏽.
But I have a question, why does it still work on android leaving the variable to false?

mikolak, 2020-07-03:
To be honest I have no idea. 😉

Write with response is the default on Android, so perhaps it falls back to it? 

@dariuszseweryn perhaps you know?

closed 2020-07-03 by mikolak

dariuszseweryn, 2020-07-03:
Never tried using write that was not supported by a characteristic. Android may use the default or because of some bug in implementation may send a write without response anyway. Or there may be a misshandling in the library itself which did correctly set "no response"?