dotintent/FlutterBleLib #493: requestMtu Back to Default

arizki29, 2020-07-16:
i have tried set mtu but value always back to default **mtu(23)**.

flutter_ble_lib: ^2.2.3

this my code:

await peripheral.connect(requestMtu: 0);
await peripheral.discoverAllServicesAndCharacteristics();
int mtu = await peripheral.requestMtu(512);

result : 
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 7491): on native side observed method: characteristicsForService
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 7491): on native side observed method: requestMtu
D/com.polidea.flutter_ble_lib.delegate.MtuDelegate( 7491): **Request MTU 512**
D/BluetoothGatt( 7491): configureMTU() - device: 4C:24:98:77:6F:B1 **mtu: 512**
**D/BluetoothGatt( 7491): onConfigureMTU() - Device=4C:24:98:77:6F:B1 mtu=23 status=0**

mikolak, 2020-07-16:
I'd guess your peripheral doesn't support higher MTU.

arizki29, 2020-07-20:
> I'd guess your peripheral doesn't support higher MTU.

sory for bad my english.
I tried Samsung J7 Prime with Android 8.0, Samsung Galaxy A01 with Android 10. and Xiaomi Redmi 3. 
and to ensure that I try to install the application "Serial Bluetooth Serial (support BlE)" on the PlayStore and the result is that data can be received completely. 
I do not know what I missed in this example FlutterBleLib.
thx

mikolak, 2021-04-07:
I don't know what might be wrong, your code looks ok. Perhaps it's some kind of weird firmware issue?

Closing because of lack of follow ups.

closed 2021-04-07 by mikolak