dotintent/FlutterBleLib #504: [Question] Scan for a specific device category

tuc0w, 2020-07-29:
Hey everyone,

is it possible to only scan for a specific device type like other smartphones or smartwatches? If not, is it possible to get the type of a scanned device?

Thanks a lot!

mikolak, 2020-08-01:
Hi!

As far as I know BLE is not aware of such things. The best you can do is parse manufacturer's data or check advertised services UUIDs. The application turning the device into a peripheral decides on the contents of both.

tuc0w, 2020-08-02:
Thank you for the advise that would be my plan b, I just thought it could be possible since android etc. will also display of what type the discovered devices are.

mikolak, 2020-08-02:
You mean the Bluetooth system settings and Bluetooth Classic devices?

tuc0w, 2020-08-03:
Exactly, like in this screenshot:
[image]

In there the system does show the type of devices that are available for pairing.

mikolak, 2020-08-03:
System aggregates both classic BT and BLE devices. You can recognise BT devices based on what profiles they support, BLE probably only have the Bluetooth icon. 
Not sure how it works with smart watches, but there are official manufacturer ids, so perhaps they're just matched to that.

@dariuszseweryn perhaps you know something more?