dotintent/FlutterBleLib #505: stop scan when app goes to background

albrownwood, 2020-08-01:
I am trying to stop the ble scanning when the app goes to background using WidgetsBindingObserver to obtain the app lifecycle events

```
class _Pairing0PageState extends State<Pairing0Page> with WidgetsBindingObserver{

  @override
  void initState() {
    super.initState();
    context.read<BleBloc>().startScan();
    WidgetsBinding.instance.addObserver(this);
  }

  @override
  void dispose() {
    WidgetsBinding.instance.removeObserver(this);
    super.dispose();
  }

  @override
  void didChangeAppLifecycleState(AppLifecycleState state) {
    print('didChangeAppLifecycleState');
    if(state == AppLifecycleState.resumed){
      context.read<BleBloc>().startScan();
    }
    else{
      context.read<BleBloc>().stopScan();
    }
  }
...
}
```

when the app goes to background, the scan stops (and correctly resumes when app comes foreground) but I get the following error message:
```
E/flutter (30336): [ERROR:flutter/lib/ui/ui_dart_state.cc(157)] Unhandled Exception: PlatformException(error, Attempt to invoke interface method 'void com.polidea.multiplatformbleadapter.BleAdapter.stopDeviceScan()' on a null object reference, null)
E/flutter (30336): #0      StandardMethodCodec.decodeEnvelope (package:flutter/src/services/message_codecs.dart:569:7)
E/flutter (30336): #1      MethodChannel._invokeMethod (package:flutter/src/services/platform_channel.dart:156:18)
E/flutter (30336): <asynchronous suspension>
E/flutter (30336): #2      MethodChannel.invokeMethod (package:flutter/src/services/platform_channel.dart:329:12)
E/flutter (30336): #3      ScanningMixin.stopDeviceScan (package:flutter_ble_lib/src/bridge/scanning_mixin.dart:50:26)
E/flutter (30336): #4      InternalBleManager.stopPeripheralScan (package:flutter_ble_lib/src/internal_ble_manager.dart:63:48)
E/flutter (30336): #5      BleBloc.stopScan (package:yunga/blocs/ble_bloc.dart:53:23)
E/flutter (30336): #6      _Pairing0PageState.didChangeAppLifecycleState (package:yunga/pages/pairing0_page.dart:38:31)
E/flutter (30336): #7      WidgetsBinding.handleAppLifecycleStateChanged (package:flutter/src/widgets/binding.dart:655:16)
E/flutter (30336): #8      SchedulerBinding._handleLifecycleMessage (package:flutter/src/scheduler/binding.dart:349:5)
E/flutter (30336): #9      BasicMessageChannel.setMessageHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:74:49)
E/flutter (30336): #10     _DefaultBinaryMessenger.handlePlatformMessage (package:flutter/src/services/binding.dart:199:33)
E/flutter (30336): #11     _invoke3.<anonymous closure> (dart:ui/hooks.dart:290:15)
E/flutter (30336): #12     _rootRun (dart:async/zone.dart:1184:13)
E/flutter (30336): #13     _CustomZone.run (dart:async/zone.dart:1077:19)
E/flutter (30336): #14     _CustomZone.runGuarded (dart:async/zone.dart:979:7)
E/flutter (30336): #15     _invoke3 (dart:ui/hooks.dart:289:10)
E/flutter (30336): #16     _dispatchPlatformMessage (dart:ui/hooks.dart:164:5)
```

Am I using the right approach here ?

mikolak, 2020-08-01:
Hi!

It seems the native client has been destroyed. Are you by any chance calling `bleManager.destroyClient()` somewhere?

albrownwood, 2020-08-01:
Hello,
yes indeed, in the stopScan function I am calling 
```
await _bleManager.stopPeripheralScan();
await _bleManager.destroyClient();
```

which I thought would be fine since I call the following in startScan:
```
_bleManager = BleManager();
await _bleManager.createClient();
_bleManager.startPeripheralScan(...);
```

mikolak, 2020-08-01:
I think it might be related to https://github.com/Polidea/FlutterBleLib/issues/503
We've recently introduced additional call to stopScan when unsubsribing from peripheral scan, which is probably called after going to background (and destroying the client beforehand). 
If that's the case, possible workarounds for now would be:
- unsubscribing from the stream instead of stopping the scan
- avoiding destroying the client.

Meanwhile I'll try to figure out how to fix it.

albrownwood, 2020-08-01:
Thanks, I took the temporary workaround of not destroying the client when app goes background for now.
Keeping this open.

VinciShark, 2020-08-14:
@mikolak Thanks for your excellent package. Have you have a time plan for this issue, I just checked in Project this is not in TO DO column

mikolak, 2020-08-14:
Hi! I'm afraid I don't have the time lately and I'm unable to tell for how long will this situation go on.

michael-ottink, 2021-01-22:
Any update on this?

dannyalan, 2021-10-31:
+1 and following