dotintent/FlutterBleLib #513: scan get empty result after shut down Bluetooth Adapter during last time scanning

VinciShark, 2020-08-25:
......
- start scan
- successfully get scan result
- shut down Bluetooth Adapter manualy
- open Bluetooth Adapter using  await BleManager.enable();
- NO MATTER whether I use await BleManager.stopPeripheralScan();
- start scan again
- no result

VinciShark, 2020-08-25:
```
2020-08-25 18:10:21.304 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: createClient
2020-08-25 18:10:21.591 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: setLogLevel
2020-08-25 18:10:21.593 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2020-08-25 18:10:21.603 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2020-08-25 18:10:29.347 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2020-08-25 18:10:30.491 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: startDeviceScan
2020-08-25 18:10:37.169 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2020-08-25 18:10:37.325 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
2020-08-25 18:10:37.327 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2020-08-25 18:10:40.156 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: getState
2020-08-25 18:10:40.163 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: enableRadio
2020-08-25 18:10:43.778 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: startDeviceScan
2020-08-25 18:10:43.790 20225-20225/com.racehf.app D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: stopDeviceScan
```

VinciShark, 2020-08-25:
after I cancel the no result scan, and start scan again, everything OK.
but no use adding BleManager.stopPeripheralScan(); after BleManager.enable();

VinciShark, 2020-08-25:
@mikolak Would you please helping me understand this, thanks

mikolak, 2020-08-25:
There's a race between starting scan and emitting the first error. If the error has occurred on start of the scan, it's possible that the subscription (and the subsequent creation of event channel in the native layer) will be done _after_ the first error has been emitted (it just hit void listener, so is not propagated). We've tried to fix this already, but it seems we need a different solution for this.

For now I think the best workaround would be to add a timeout condition that would trigger cancelling of the scan if no results have been found in a few seconds.

VinciShark, 2020-08-25:
Thanks for your workaround advice.

Another question, is this issue, or are the most part of issues under this repo exist in exactly this repo(FlutterBleLib), or AndroidBleLib?
I am considering if I have time and ability to be part of this repo's contributor.

mikolak, 2020-08-25:
Do you mean MultiPlatformBleAdapter or RxAndroidBle by AndroidBleLib? Issues concerning this library are mostly here, some might show up on MultiPlatformBleAdapter or be duplicated in react-native-ble-plx (as it now uses the same backend, MultiPlatformBleAdapter).