dotintent/FlutterBleLib #514: can't read from readable Characteristic

Matrix-Zhang, 2020-08-25:
```
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin( 7591): on native side observed method: readCharacteristicForIdentifier
E/flutter ( 7591): [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: BleError (Error code: 2, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
E/flutter ( 7591): 

```

mikolak, 2020-08-25:
Error code 2 means operation cancelled. I think you might've passed the same transaction ID to two operations.

Matrix-Zhang, 2020-08-26:
> Error code 2 means operation cancelled. I think you might've passed the same transaction ID to two operations.

Okay, i think the transaction id is session id....

after i remove the transaction id, new error comes
```
E/flutter (24967): [ERROR:flutter/lib/ui/ui_dart_state.cc(166)] Unhandled Exception: BleError (Error code: 3, ATT error code: null, iOS error code: null, Android error code: null, reason: GATT exception from MAC address 5B:14:86:24:98:64, with type BleGattOperation{description='CHARACTERISTIC_READ'}, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
```

mikolak, 2020-08-26:
3 is operation timeout. https://github.com/Polidea/FlutterBleLib/blob/develop/lib/error/ble_error.dart

This library is based on our React Native BLE library and some things are directly translated from there. You can check out what the errors mean here: https://polidea.github.io/react-native-ble-plx/#bleerrorcode

I need to add docs to those error codes here...

Matrix-Zhang, 2020-08-26:
> 3 is operation timeout. https://github.com/Polidea/FlutterBleLib/blob/develop/lib/error/ble_error.dart
>
> This library is based on our React Native BLE library and some things are directly translated [...]

why timeout? if i remove the Characteristic's read method, just use monitor, the data will received....

mikolak, 2020-08-26:
I don't why, I haven't seen your code. 🙂 If you're not monitoring, does the read succeed?

Matrix-Zhang, 2020-08-27:
> I don't why, I haven't seen your code. 🙂 If you're not monitoring, does the read succeed?

if i not monitoring, the read timeout,

the code just like:
```
write("1234") ; //send to peripheral
read(); // read from peripheral timeout
```

Matrix-Zhang, 2020-08-31:
if i read from ios, the error code is 402

mikolak, 2020-09-16:
Can you check the permissions on your characteristic? (`isReadable`, `isWritable`, etc.)

Matrix-Zhang, 2020-09-21:
> Can you check the permissions on your characteristic? (`isReadable`, `isWritable`, etc.)

yes, i have checked it already, the characteristic is readable