dotintent/FlutterBleLib #515: Accessing hidden methods warnings on Android 10+

b055man, 2020-08-26:
When using the plugin on Android 10, there are number of warnings issued due to using non-public APIs.

Are there any plans on resolving this (Google at some point may become more restrictive..)?

`
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(14382): on native side observed method: isDeviceConnected
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(14382): on native side observed method: connectToDevice
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->objectFieldOffset(Ljava/lang/reflect/Field;)J (greylist,core-platform-api, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->compareAndSwapInt(Ljava/lang/Object;JII)Z (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->getIntVolatile(Ljava/lang/Object;J)I (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden field Lsun/misc/Unsafe;->theUnsafe:Lsun/misc/Unsafe; (greylist, reflection, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->arrayIndexScale(Ljava/lang/Class;)I (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->arrayBaseOffset(Ljava/lang/Class;)I (greylist,core-platform-api, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->getObject(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->getLongVolatile(Ljava/lang/Object;J)J (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedLong(Ljava/lang/Object;JJ)V (greylist, linking, allowed)
W/ra.smartcontro(14382): Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)`

mikolak, 2020-08-28:
@dariuszseweryn could it be from RxAndroidBle or is it from MBA?

dariuszseweryn, 2020-08-28:
Nothing in the `RxAndroidBle` is directly referencing `sun.misc.Unsafe` class. There are some reflection use-cases that workaround connection issues on older Android versions. They can be viewed [here](https://github.com/Polidea/RxAndroidBle/blob/master-rxjava1/rxandroidble/src/main/java/com/polidea/rxandroidble/internal/util/BleConnectionCompat.java). There is nothing else that I know of.

b055man, 2020-08-28:
I think this is exactly related to reflections. Here is a snippet from the pre-launch report from Play Console:
[image]

This is from a full app I'm working on, so it is possible that not all of these apply to flutter_ble, but I believe at least some of them do, as they resemble the warnings I observe in the console when there are BLE operations happening.
[image]

Any chance to have an SDK-dependant implementation of things in the future that do not use hidden APIs on modern SDKs?