dotintent/FlutterBleLib #522: [Flutter: ScanResult library peripheral scan return Null]

Rayen-cherni, 2020-09-10:
I have a flutter application where I need to use Bluetooth Low Energy using this Library and I need to check the name of other devices:
Library :
https://pub.dev/packages/flutter_ble_lib
Code :
`...`
`ScanResult scanResult`
`...`
`print("Scanned Peripheral ${scanResult.peripheral.name}")`
Result :
Scanned Peripheral null

mikolak, 2020-09-10:
Both local name and name are nullable, they may not exist on a peripheral.

closed 2020-09-10 by mikolak

Rayen-cherni, 2020-09-10:
So how can i get device's names ?

mikolak, 2020-09-11:
You check those two fields. If your device is certain to have a name, it will be on one of those. If the device doesn't have a name, then it doesn't have a name, nothing you could do about it.