dotintent/FlutterBleLib #523: Exception on iOS after relaunch

tmorone-rezi, 2020-09-14:
I am experiencing an exception on iOS that sometimes occurs if I shutdown my app, relaunch, and then attempt a scan. I also see this (sometimes) if I perform a scan, back out of the screen, and then perform the scan again.

On the call to startPeripheralScan I get this error:
Exception has occurred.
BleError (BleError (Error code: 103, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null))

My environment:
iPhone 11
iOS 13.7
xcode 11.7
flutter 1.20.3
flutter_ble_lib 2.3.0

mikolak, 2020-09-16:
103 is `Bluetooth in unknown state`. It might be a race condition in your code. 
Can you try adding observation of adapter state? `bleManager.observeBluetoothState(emitCurrentState: true)`

tmorone-rezi, 2020-09-16:
Thank you! That made a huge difference. This will wait for POWERED_ON before doing the peripheral scan:

await bleManager.createClient();
await for (BluetoothState state in bleManager.observeBluetoothState()) {
  if (state == BluetoothState.POWERED_ON) {
    break;
  }}

closed 2020-09-16 by tmorone-rezi

sejunoh, 2021-09-28:
> Thank you! That made a huge difference. This will wait for POWERED_ON before doing the peripheral scan:
> 
> await bleManager.createClient();
> await for (BluetoothState state in bleManager.observeBluetoothState()) {
> if (state == BluetoothState.POWERED_ON) {
> break;
> }}

Great! It works for me.