dotintent/FlutterBleLib #524: Could not write to the right characteristic which is present in multiple services

NailAgliev, 2020-09-16:
I have two different services with different UUIDs that have characteristics with the same UUIDs.
Write to the first service characteristic everything works as expected.
Trying to write to second service characteristic results in data will be received in the first characteristic and most of the time data is corrupted.

NailAgliev, 2020-09-16:
I'll check ios and suggestions tomorrow

NailAgliev, 2020-09-17:
`peripheral.services().firstWhere(<your service>).characteristics().firstWhere(<your characteristic>)` provides the same result.

There is logcat output

    -------------------------------------------------------------------------
    
    Primary Service - Generic Attribute (00001801-0000-1000-8000-00805f9b34fb)
    Instance ID: 1
    -> Characteristics:
    	* Service Changed (00002a05-0000-1000-8000-00805f9b34fb)
    	  Properties: [ INDICATE ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Generic Access (00001800-0000-1000-8000-00805f9b34fb)
    Instance ID: 20
    -> Characteristics:
    	* Device Name (00002a00-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    	* Appearance (00002a01-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    	* Central Address Resolution (00002aa6-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ ]
    
    Primary Service - Weight Scale (0000181d-0000-1000-8000-00805f9b34fb)
    Instance ID: 40
    -> Characteristics:
    	* Weight (00002a98-0000-1000-8000-00805f9b34fb)
    	  Properties: [ READ NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Unknown service (db3e25d0-79e9-48fa-a2c3-facc90a3d437)
    Instance ID: 55
    -> Characteristics:
    	* Unknown characteristic (db3e25d1-79e9-48fa-a2c3-facc90a3d437)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Unknown service (db3e25d6-79e9-48fa-a2c3-facc90a3d437)
    Instance ID: 70
    -> Characteristics:
    	* Unknown characteristic (db3e25d7-79e9-48fa-a2c3-facc90a3d437)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Unknown service (6e400001-b5a3-f393-e0a9-e50e24dcca9e)
    Instance ID: 85
    -> Characteristics:
    	* Unknown characteristic (6e400003-b5a3-f393-e0a9-e50e24dcca9e)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    	* Unknown characteristic (db3e25d3-79e9-48fa-a2c3-facc90a3d437)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    	* Unknown characteristic (db3e25d4-79e9-48fa-a2c3-facc90a3d437)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
    		* Client Characteristic Configuration (00002902-0000-1000-8000-00805f9b34fb)
    
    Primary Service - Unknown service (db3e25d5-79e9-48fa-a2c3-facc90a3d437)
    Instance ID: 100
    -> Characteristics:
    	* Unknown characteristic (db3e25d1-79e9-48fa-a2c3-facc90a3d437)
    	  Properties: [ WRITE NOTIFY ]
    	  -> Descriptors: 
[... 15 lines cut ...]
    D/BluetoothGatt: setCharacteristicNotification() - uuid: 00002a98-0000-1000-8000-00805f9b34fb enable: true
    W/u.flutter_scal: Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
    D/RxBle#ConnectionOperationQueue: QUEUED   DescriptorWriteOperation(8831607)
    D/RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
    W/u.flutter_scal: Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
    W/u.flutter_scal: Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
    D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(2411625) in 240 ms
    D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(16110586)
    D/RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=db3e25d1-79e9-48fa-a2c3-facc90a3d437 status=0
    V/RxBle#BleModule: Write to Characteristic(uuid: db3e25d1-79e9-48fa-a2c3-facc90a3d437, id: 12, value: 56)
    D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(16110586) in 7 ms
    D/RxBle#ConnectionOperationQueue: STARTED  DescriptorWriteOperation(8831607)
    W/u.flutter_scal: Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
    D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: services
    D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: characteristicsForService
    I/flutter: write to 2nd
    D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin: on native side observed method: writeCharacteristicForIdentifier
    D/RxBle#ConnectionOperationQueue: QUEUED   CharacteristicWriteOperation(254389343)
    D/RxBle#BluetoothGatt$1: onDescriptorWrite descriptor=00002902-0000-1000-8000-00805f9b34fb status=0
    W/u.flutter_scal: Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
    D/RxBle#ConnectionOperationQueue: FINISHED DescriptorWriteOperation(8831607) in 89 ms
    D/RxBle#ConnectionOperationQueue: STARTED  CharacteristicWriteOperation(254389343)
    D/RxBle#BluetoothGatt$1: onCharacteristicWrite characteristic=db3e25d1-79e9-48fa-a2c3-facc90a3d437 status=0
    V/RxBle#BleModule: Write to Characteristic(uuid: db3e25d1-79e9-48fa-a2c3-facc90a3d437, id: 25, value: 56)
    D/RxBle#ConnectionOperationQueue: FINISHED CharacteristicWriteOperation(254389343) in 5 ms

NailAgliev, 2020-12-12:
Any new information?

mikolak, 2021-04-12:
OMG, sorry for the wait. Do you by any chance still have the faulty code/device?