dotintent/FlutterBleLib #527: Natenczas Wojski

JasonEdinburgh, 2020-09-17:
When I use NRFconnect I see that the server has these fields

[image]

First Name "Natenczas"
Last Name "Wojski"

He seems to be one of the characters in Adam Mickiewicz's epic Pan Tadeusz . 

**Is there any way for me to change these fields?**

mikolak, 2020-09-17:
Hi!

Our library handles the `client` tab from nRF Connect. Are you sure you've posted this in the right place? 🙂 

I don't know what peripheral you're using, but I'm guessing that since someone set them, you could that as well.

closed 2020-09-17 by mikolak

JasonEdinburgh, 2020-09-17:
Sorry if I didn't make myself clear. The device was connected to my flutter app with FlutterBleLib. I assumed that this is why the values were shown in the server tab.

JasonEdinburgh, 2020-09-17:
you are right..... this is not related to FlutterBleLib. I see this without connecting with my phone..

How strange! Sorry for wasting your time

Xprogrammer777, 2022-07-22:
Location: chest.. etc... look like l8nux sarcastic humor I don't understand... It is i a joke from RFconnect?