dotintent/FlutterBleLib #531: discoverAllServicesAndCharacteristics() never returns on iOS

KaeferUndCoins, 2020-10-02:
I tried enabling verbose logging and checked the XCode output, but there is no indication for any errors.
Last related output with the peripheral's UUID is

```Peripheral(uuid: [DEVICE UUID], name: Optional("[Device Name]")) discoverServices(
serviceUUIDs: nil)```

Next up is the error when I timeout the operation and disconnect from the application level.

Android working just fine though...

Lootwig, 2020-10-02:
I tried enabling verbose logging and checked the XCode output, but there is no indication for any errors.
Last related output with the peripheral's UUID is

```Peripheral(uuid: [DEVICE UUID], name: Optional("[Device Name]")) discoverServices(serviceUUIDs: nil)```

Next up is the error when I timeout the operation and disconnect from the application level.

Android working just fine though...

mikolak, 2020-10-02:
Could you check doing the same using nRF Connect? It's a free app from Nordic Semiconductors available on Appstore and Google Play.

Lootwig, 2020-10-03:
That's super weird. iOS nRF doesn't detect the service I'm expecting either. Might be an issue with the device, it's a prototype - I'll have this checked out by the manufacturer.

The nRF app _does_ however finish service discovery, I'm not familiar enough with the matter to say if this is just due to an application-side timeout or an actual difference in the handling of a (maybe?) corrupted data format etc. that doesn't seem to cause problems on Android?

I will post an update on the missing service as soon as I confirmed the issue with the manufacturer!

mikolak, 2020-10-04:
nRF Connect allows you to initiate bonding with the device. Could you try that and see if the service shows up?

Lootwig, 2020-10-07:
This only works in the Android version. I checked why nRF on iOS doesn't offer the option and found this article:
[https://devzone.nordicsemi.com/f/nordic-q-a/48997/nrf-connect-app-in-ios-don-t-have-bond-option](url)

I'm not sure I get the bonding step though, the Ble Lib does not have a function for bonding in the API, how would I even do that using the library?

closed 2021-02-22 by KaeferUndCoins