dotintent/FlutterBleLib #533: non-SDK interface error when reading and writing

joshting, 2020-10-11:
I hope this is something not related to the library but some thing I can configure to make it go away.  I narrow down the problem to each time I perform read or write to a charateristic.

In the android app build.gradle: minSdkVersion is 19 and targetSdkVersion is 28

Accessing hidden method Lsun/misc/Unsafe;->getObjectVolatile(Ljava/lang/Object;J)Ljava/lang/Object; (greylist, linking, allowed)
Accessing hidden method Lsun/misc/Unsafe;->putOrderedObject(Ljava/lang/Object;JLjava/lang/Object;)V (greylist, linking, allowed)
Accessing hidden method Lsun/misc/Unsafe;->getLongVolatile(Ljava/lang/Object;J)J (greylist, linking, allowed)

mikolak, 2021-04-12:
Wait, so what's the error? These are warnings, underlying native code has to be ported to newer API, but everything should be working fine at the moment.

mikolak, 2021-04-12:
Duplicates #515

closed 2021-04-12 by mikolak