dotintent/FlutterBleLib #534: Cannot see Android Devices in the list

emretufekci, 2020-10-12:
Hello,

I have two android mobiles, and i'm using Example app of "FluuterBleLib" 

When i'm running the app, it can list Smartwatches, iphones but no android phone.

I would like to communicate these two android mobile phones each other but i cannot.

As i know theoretically it should be possible, after Android 5+ the device can behave like peripheral & central at the same time.

I would like to make sure  is it hardware issue or software..

I'm using Android 9+

mikolak, 2020-10-12:
Hi! The library does not support peripheral mode at the moment.

There's this https://pub.dev/packages/flutter_ble_peripheral, haven't used it.

emretufekci, 2020-10-12:
So do you plan to implement it? Thank you so much for quick reply! I really appreciate it! :1st_place_medal:

emretufekci, 2020-10-12:
Dear, @mikolak I used the library which is you shared, to make phone as peripheral. 

So with the other phone (central one) i've tried to list but i couldn't list. I've change the library (BleLib to Other one) and it can list peripheral phone.

So, i'm just reporting this issue to make you aware about the problem.

mikolak, 2020-10-12:
> So do you plan to implement it? Thank you so much for quick reply! I really appreciate it! :1st_place_medal: 

We'd like to have it one day, yes, but I cannot tell if we will. 

> Dear, @mikolak I used the library which is you shared, to make phone as peripheral.
>
> So with the other phone (central one) i've tried to list but i couldn't list. I've change the library (BleLib to [...]

Was there any filtering turned on? The library's example filters peripherals by name (it's set in devices_list_bloc IIRC).

emretufekci, 2020-10-12:
> 
> Was there any filtering turned on? The library's example filters peripherals by name (it's set in devices_list_bloc IIRC).

I haven't touch any piece of code. I've just used the example one. But as i can see with other library it is lists with name of the device (peripheral phone's name).

mikolak, 2020-10-12:
Line 113
https://www.github.com/Polidea/FlutterBleLib/tree/develop/example%2Flib%2Fdevices_list%2Fdevices_bloc.dart

If you remove the condition that local name is not null, the device should be visible.

emretufekci, 2020-10-13:
Yes correct after that it's working.

I guess the library which is you shared to make phone as peripheral does not advertise any data,  so for the prevent null object in the list line 113 should be like this:

      _if (scanResult.peripheral.name != null &&_

Mr. @mikolak, Thank you for effort, you're really helpful...

closed 2020-10-13 by emretufekci