dotintent/FlutterBleLib #541: Is  full-duplex communication possible

emretufekci, 2020-11-10:


mikolak, 2020-11-10:
The only way to do that would be to create a peripheral module in the app and have your BLE device act not only as the peripheral, but also a central. This library only supports central role at the moment, so you'd need something like https://pub.dev/packages/flutter_ble_peripheral

If you're not familiar with those terms, you can think of peripheral role as a server and the central as a client.

JasonEdinburgh, 2020-11-10:
I may be misunderstanding this conversation, but I have 2 way communication. 

I use the Nordic uart service on my NRF52840 based device and FlutterBleLib on the smartphone.

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.0.0%2Fble_sdk_app_nus_eval.html

emretufekci, 2020-11-10:
@JasonEdinburgh Hi Jason, thanks for answer. 
I have two way communication as well but not via gatt server on both side.

For ex:
Android (Gatt Client) -> BLE (Gatt Server) [NOTIFICATION ENABLED]
_By doing this i can receive all datas from ble device_

Android (Gatt Client) -> BLE(Gatt Server) [WRITE WITHOUT RESPONSE]
_By doing this i can send all datas to ble device_

Instead of doing this, i would like to create a notification service on mobile phone side (so actually it is making phone as server), so ble device can enabled my notification then i can use full duplex communication.

So 

_RX Characteristic (UUID: 6E400002-B5A3-F393-E0A9-E50E24DCCA9E)
The peer can send data to the device by writing to the RX Characteristic of the service. ATT Write Request or ATT Write Command can be used. The received data is sent on the UART interface._

_TX Characteristic (UUID: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E)
If the peer has enabled notifications for the TX Characteristic, the application can send data to the peer as notifications. The application will transmit all data received over UART as notifications._

As far as i understood; 

This is offering the same way what i have currently.

emretufekci, 2020-11-10:
@mikolak Thanks for advise, the library seems missing features. I will test it one more time. Thanks again.

JasonEdinburgh, 2020-11-10:
yes, sorry, I misunderstood.

closed 2020-11-18 by mikolak