dotintent/FlutterBleLib #543: iOS XPC Connection Invalid

jtrexr2, 2020-11-12:
Hello,

I have a flutter app sending configurations over Bluetooth to esp32-based devices; I'm able to connect and send over multiple packets, but then once the connection completes and I return to scanning and try to connect to other devices, I get `[CoreBluetooth] XPC connection invalid`, and a Bluetooth state of UNSUPPORTED. 

I'm able to connect, transmit, disconnect, scan, connect to another device, etc without issue on Android.

I would appreciate any help you could provide! A more full log below:

```
2020-11-11 21:36:51.144505-0500 Runner[26518:1919800] [CoreBluetooth] XPC connection invalid
2020-11-11 21:36:51.185549-0500 Runner[26518:1919807] flutter: Ble client created
2020-11-11 21:36:51.188484-0500 Runner[26518:1919807] [VERBOSE-2:ui_dart_state.cc(177)] Unhandled Exception: BleError (Error code: 100, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
#0      ScanningMixin.startDeviceScan.<anonymous closure> (package:flutter_ble_lib/src/bridge/scanning_mixin.dart:33:9)
#1      _invokeErrorHandler (dart:async/async_error.dart:16:24)
#2      _HandleErrorStream._handleError (dart:async/stream_pipe.dart:282:9)
#3      _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:161:13)
#4      _rootRunBinary (dart:async/zone.dart:1214:47)
#5      _CustomZone.runBinary (dart:async/zone.dart:1107:19)
#6      _CustomZone.runBinaryGuarded (dart:async/zone.dart:1013:7)
#7      _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:376:15)
#8      _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:394:16)
#9      _DelayedError.perform (dart:async/stream_impl.dart:622:14)
#10     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:730:11)
#11     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:687:7)
#12     _rootRun (dart:async/zone.dart:1182:47)
#13     _CustomZone.run (dart:async/zone.dart:1093:19)
#14     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#15     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#16     _rootRun (dart:async/zone.dart:1190:13)
#17     _CustomZone.run (dart:async/zone.dart:1093:19)
#18     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#19     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#20     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
#21     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
2020-11-11 21:36:51.189624-0500 Runner[26518:1919807] [VERBOSE-2:ui_dart_state.cc(177)] Unhandled Exception: BleError (Error code: 100, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)
#0      ScanningMixin.startDeviceScan.<anonymous closure> (package:flutter_ble_lib/src/bridge/scanning_mixin.dart:33:9)
#1      _invokeErrorHandler (dart:async/async_error.dart:16:24)
#2      _HandleErrorStream._handleError (dart:async/stream_pipe.dart:282:9)
#3      _ForwardingStreamSubscription._handleError (dart:async/stream_pipe.dart:161:13)
#4      _rootRunBinary (dart:async/zone.dart:1214:47)
#5      _CustomZone.runBinary (dart:async/zone.dart:1107:19)
#6      _CustomZone.runBinaryGuarded (dart:async/zone.dart:1013:7)
#7      _BufferingStreamSubscription._sendError.sendError (dart:async/stream_impl.dart:376:15)
#8      _BufferingStreamSubscription._sendError (dart:async/stream_impl.dart:394:16)
#9      _DelayedError.perform (dart:async/stream_impl.dart:622:14)
#10     _StreamImplEvents.handleNext (dart:async/stream_impl.dart:730:11)
#11     _PendingEvents.schedule.<anonymous closure> (dart:async/stream_impl.dart:687:7)
#12     _rootRun (dart:async/zone.dart:1182:47)
#13     _CustomZone.run (dart:async/zone.dart:1093:19)
#14     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#15     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#16     _rootRun (dart:async/zone.dart:1190:13)
#17     _CustomZone.run (dart:async/zone.dart:1093:19)
#18     _CustomZone.runGuarded (dart:async/zone.dart:997:7)
#19     _CustomZone.bindCallbackGuarded.<anonymous closure> (dart:async/zone.dart:1037:23)
#20     _microtaskLoop (dart:async/schedule_microtask.dart:41:21)
#21     _startMicrotaskLoop (dart:async/schedule_microtask.dart:50:5)
2020-11-11 21:37:05.957595-0500 Runner[26518:1919807] flutter: Bluetooth state: BluetoothState.UNSUPPORTED
2020-11-11 21:37:05.958331-0500 Runner[26518:1919807] flutter: Ble client created
2020-11-11 21:37:11.948900-0500 Runner[26518:1919807] flutter: Bluetooth state: BluetoothState.UNSUPPORTED
```

andreibratuleanu21, 2021-05-19:
Same issue here on iOS 13.3

On iOS 12.x it's working fine.

andreibratuleanu21, 2021-05-27:
There is this `restoreStateIdentifier` argument that you can feed into `createClient()` method.
Seems that for this lib to work on iOS 14.x this is required. It's working for me now.

https://github.com/Polidea/FlutterBleLib/blob/develop/lib/ble_manager.dart#L69

Maarsls, 2021-08-05:
Hey,
is there any solution or @andreibratuleanu21 could you please explain me further how to fix that !