dotintent/FlutterBleLib #549: Error Status 402 - Not reading characteristic

a8thunder, 2020-11-24:
### **Hi , I connected the Device but not reading the characteristic . AnyBody can help me pls. Thanks you**
`await peripheral
        .discoverAllServicesAndCharacteristics()
        .then((_) => peripheral.services())
        .then((services) {
      log('PRINTING SERVICES for ${peripheral.name}');
      return services.first;
    }).then((service) async {
      log('PRINTING CHARACTERISTICS FOR SERVICE \n${service.uuid}');
      List<Characteristic> characteristics = await service.characteristics();
      log('PRINTING CHARACTERISTICS FROM \nPERIPHERAL for the same service');
      return peripheral.characteristics(service.uuid);
    }).then((characteristics) {
      peripheral.requestMtu(185);
      for (Characteristic characteristic in characteristics) {
        log('Found characteristic \n ${characteristic.uuid}');
        characteristic.isReadable = true;
        characteristic.isWritableWithResponse = true;
        characteristic.isWritableWithoutResponse = true;
        if (characteristic == null) {
          print("Characteristic not found");
        } else {
          print(
            "Characteristic:\n"
            "uuid: ${characteristic.uuid}\n"
            "isReadable: ${characteristic.isReadable}\n"
            "isWritableWithoutResponse: ${characteristic.isWritableWithoutResponse}\n"
            "isWritableWithResponse: ${characteristic.isWritableWithResponse}\n"
            "isIndicatable: ${characteristic.isIndicatable}\n"
            "isNotifiable: ${characteristic.isNotifiable}\n",
          );
        }
        Stream<CharacteristicWithValue> characteristicUpdates =
            peripheral.monitorCharacteristic(
                '000018f0-0000-1000-8000-00805f9b34fb',
                '00002af0-0000-1000-8000-00805f9b34fb');
        characteristicUpdates.listen((event) {
          print('##################');
          print('$event');
          readCharacteristic();
        }, onError: (error) {
          print('##################');
          print('$error');
        }).resume();
        if (characteristic.uuid.toString().contains(writeCharacteristicUUID)) {
          Future<void>.delayed(const Duration(milliseconds: 500), () async {
            // characteristic.is
            try {
              print(Uint8List.fromList(utf8.encode('ATZ\n')));
              // await characteristic.write(
              //     Uint8List.fromList(utf8.encode('ATZ\n')), false);
              await peripheral.writeCharacteristic(
                  '000018f0-0000-1000-8000-00805f9b34fb',
                  '00002af1-0000-1000-8000-00805f9b34fb',
                  Uint8List.fromList(utf8.encode('ATZ\n')),
                  false);
            } catch (e) {
              print(e.toString());
            }
          });
        }
      }
    });`

**this is my console:**
flutter: Characteristic{service: Service{peripheralId: F2CCD470-E0F6-2D73-9495-09D446F6DC08, uuid: 000018f0-0000-1000-8000-00805f9b34fb}, _manager: Instance of 'InternalBleManager', uuid: 00002af0-0000-1000-8000-00805f9b34fb, isReadable: false, isWritableWithResponse: false, isWritableWithoutResponse: false, isNotifiable: true, isIndicatable: true} CharacteristicWithValue{value = [65, 84, 90, 10] CharacteristicWithValueAndTransactionId{transactionId: 3}
flutter: Reading temperature config
flutter: ##################
flutter: Characteristic{service: Service{peripheralId: F2CCD470-E0F6-2D73-9495-09D446F6DC08, uuid: 000018f0-0000-1000-8000-00805f9b34fb}, _manager: Instance of 'InternalBleManager', uuid: 00002af0-0000-1000-8000-00805f9b34fb, isReadable: false, isWritableWithResponse: false, isWritableWithoutResponse: false, isNotifiable: true, isIndicatable: true} CharacteristicWithValue{value = [65, 84, 90, 10] CharacteristicWithValueAndTransactionId{transactionId: 4}
flutter: Reading temperature config
flutter: ##################
flutter: BleError (Error code: 402, ATT error code: 2, iOS error code: null, Android error code: null, reason: Reading is not permitted., internal message: null, device ID: F2CCD470-E0F6-2D73-9495-09D446F6DC08, service UUID: 000018f0-0000-1000-8000-00805f9b34fb, characteristic UUID: 00002af0-0000-1000-8000-00805f9b34fb, descriptor UUID: null)
flutter: ##################
flutter: BleError (Error code: 402, ATT error code: 2, iOS error code: null, Android error code: null, reason: Reading is not permitted., internal message: null, device ID: F2CCD470-E0F6-2D73-9495-09D446F6DC08, service UUID: 000018f0-0000-1000-8000-00805f9b34fb, characteristic UUID: 00002af0-0000-1000-8000-00805f9b34fb, descriptor UUID: null)
>

mikolak, 2020-11-24:
The characteristic you're trying to read does not support reading, only notifications. Use `.monitor() ` and listen to the stream.

closed 2020-11-24 by mikolak

a8thunder, 2020-11-24:
Hi @mikolak 
I used monitor and check status characterstic but not read it

> uuid: 00002af0-0000-1000-8000-00805f9b34fb
> isReadable: true
> isWritableWithoutResponse: false
> isWritableWithResponse: false
> isIndicatable: true
> isNotifiable: true
> flutter: Characteristic:
> uuid: 00002af1-0000-1000-8000-00805f9b34fb
> isReadable: true
> isWritableWithoutResponse: true
> isWritableWithResponse: true
> isIndicatable: false
> isNotifiable: false
> [log] Found characteristic 
>        00002af1-0000-1000-8000-00805f9b34fb
> flutter: [65, 84, 90, 10]
> flutter: ##################
> flutter: [65, 84, 90, 10]
> [log] BleError (Error code: 402, ATT error code: 2, iOS error code: null, Android error code: null, reason: Reading is not permitted., internal message: null, device ID: F2CCD470-E0F6-2D73-9495-09D446F6DC08, service UUID: 000018f0-0000-1000-8000-00805f9b34fb, characteristic UUID: 00002af0-0000-1000-8000-00805f9b34fb, descriptor UUID: null)

for (Characteristic characteristic in characteristics) {
        log('Found characteristic \n ${characteristic.uuid}');
        // characteristic.isNotifiable = true;
        characteristic.isReadable = true;
        // characteristic.isWritableWithResponse = true;
        // // characteristic.isWritableWithoutResponse = true;
        if (characteristic == null) {
          print("Characteristic not found");
        } else {
          print(
            "Characteristic:\n"
            "uuid: ${characteristic.uuid}\n"
            "isReadable: ${characteristic.isReadable}\n"
            "isWritableWithoutResponse: ${characteristic.isWritableWithoutResponse}\n"
            "isWritableWithResponse: ${characteristic.isWritableWithResponse}\n"
            "isIndicatable: ${characteristic.isIndicatable}\n"
            "isNotifiable: ${characteristic.isNotifiable}\n",
          );
        }
        if (characteristic.uuid.toString().contains(nofityCharacteristicUUID)) {
          var monStream = characteristic.monitor();

          monStream.listen((event) {
            print('##################');
            print('$event');
            characteristic.read();
            // readCharacteristic();
          }).onError((error) {
            log(error.toString());
          });
        }
        if (characteristic.uuid.toString().contains(writeCharacteristicUUID)) {
          Future<void>.delayed(const Duration(milliseconds: 500), () async {
            print(Uint8List.fromList(utf8.encode('ATZ\n')));
            // characteristic.is
            try {
              await characteristic.write(
                  Uint8List.fromList(utf8.encode('ATZ\n')), false);
            } catch (e) {
              print(e.toString());
            }
          });
        }
      }

mikolak, 2020-11-24:
Whether a characteristic is readable or not is controlled by the peripheral. You cannot change the permission on it, you'd have to change the peripheral's firmware.

a8thunder, 2020-11-24:
Ok, thank you @mikolak