dotintent/FlutterBleLib #551: Cannot reconnect to a connected Peripheral

claudeHasler, 2020-11-30:
I am connecting successfully to a peripheral, however if my app is pushed into the background and later resumed I cannot scan for the peripheral as it is still connected.

I assumed getConnectedPeripherals() was used for this, however I am not receiving any Peripherals from this method. Should this method be working on Android?

I also tried to call destroyClient() in the hope that it would clear the surviving connection and allow the peripheral to become scannable again.

Is there any way, short of disabling and enabling the radio, to reset all active connections?

mikolak, 2020-11-30:
Hi!

You can now save the id and construct peripheral using `bleManager.createUnsafePeripheral()`. Once you have the peripheral just try to connect with catching error or call `isConnected` first.

Due to legacy reasons current implementation of `connectedDevices` returns only peripherals that were connected through this instance of the library. I'm also not sure if it is possible to retrieve all connected peripherals on the system level, but it's on our research list.

claudeHasler, 2020-12-01:
Is this a crossplatform solution or Android only? If Android only what is the iOS solution?

claudeHasler, 2020-12-01:
I can confirm that this works as described above on Android, iOS tests will be performed eventually.

What is the expected behaviour of using createUnsafePeripheral with an invalid id?

mikolak, 2020-12-01:
Should work as well on iOS. Be sure to use the timeout parameter, as it will complete the connection attempt in finite time on both platforms.