dotintent/FlutterBleLib #554: Missing Notifications / Missing Writes

claudeHasler, 2020-12-04:
My application uses two BLE Characteristics as RX/TX channels to implement a serial communication. For every write to the TX Characteristic I expect to receive an answer via notifying RX characteristic.

Generally this is working great with this library, however occaisionally (between 1000-10000) write/notify pairs I am missing a notification on the RX characteristic. I see three possibilites: The device is not answering my request, or my request is not reaching the device, or the devices answer is not reaching my application.
As I cannot (yet) debug the device I'm communication with I'd like to make sure that the error is not on my side.

Is there a way I can confirm that my flutter side writing to the TX characteristic as well as the receiving of the RX notification are not being lost on their way to/from the android/ios host?

mikolak, 2020-12-04:
Hi! You can try setting log level to verbose and then running it from the native project (xcode or Android Studio), as flutter run unfortunately does not display native logs. If your command reached radio and failed or a notification was not received, but should've been, it should be in the logs.

claudeHasler, 2020-12-07:
Thansk for the reply.
Is it enough to run the native app in Android studio and select "verbose" in the logcat menu? The output doesnt seem to be different so far. Do I need to start the app via commandline with a flag? I'm probably searching wrong but I'm not finding a guide on how to enable verbose logging anywhere.

mikolak, 2020-12-11:
Hi!

You need to use https://pub.dev/documentation/flutter_ble_lib/latest/flutter_ble_lib/BleManager/setLogLevel.html and pass it LogLeve.verbose and there should be logs from Android's Bluetooth stack.