dotintent/FlutterBleLib #556: Null-Safety migration

RTrackerDev, 2020-12-08:
Do you plan migrate to null-safety? Next Dart(2.12) will be with stable and by default null-safety.

mikolak, 2020-12-11:
Plan? Yes. When? I don't know. 🙂  Do you if that feature is backwards compatible?

Link for future reference: https://dart.dev/null-safety/migration-guide

jdar, 2020-12-25:
While there's no need a formal roadmap... it would be useful to know the "bus factor" is on this project? Is this just one engineer (you) that is maintaining this?

jlubeck, 2021-03-04:
Null safety is now on stable branch since yesterday, so it would be nice to have this so we can convert out projects to fully null safety. Thanks!

DevNico, 2021-03-12:
Would be great to have a null safe version ASAP!

okocsis, 2021-04-01:
Good news everyone:
I'm happy to announce that I've issued my very own PR to support Nullsafety.
 
I'd like you guys to have a look at my nullsafety branch:
#586
what are your thoughts? @mikolak 
(as mentioned in the pr text, please do help me with the unnecessary and or still broken tests)

thanks!

mikolak, 2021-04-02:
> While there's no need a formal roadmap... it would be useful to know the "bus factor" is on this project? Is this just one engineer (you) that is maintaining this? 

Temporarily I'm the only one on this. Had a bumpy transition and just today I've setup my machine to be able to do anything in the library. I'll try to address some of the issues next week.
There are talks with company willing to take over, but I don't know what's the state of things. 

> Good news everyone:
> I'm happy to announce that I've issued my very own PR to support Nullsafety.
>
> I'd like you guys to have a look at my nullsafety branch:
> #586
> what are your thoughts? @mikolak [...]

I'll take a look at this next week.

okocsis, 2021-04-06:
> I'll take a look at this next week.
@mikolak I've managed to fix the broken tests too.
(broken due to new nullsafe mockito API)
 I think it should be fine now! 
 Thanks for reviewing mate!
 
 @jlubeck @DevNico @jdar 
 Could you guys also have look at my branch and perhaps test if it behaves "well enough"? 
 Cheers,
 Oliver

closed 2021-04-13 by mikolak

mikolak, 2021-04-13:
FYI there's a release blocker (#591). As soon as I sort it out, I'll release 2.4.0 with null safety and API for checking if native client is allocated.

Jakegillingham5, 2021-05-26:
> FYI there's a release blocker (#591). As soon as I sort it out, I'll release 2.4.0 with null safety and API for checking if native client is allocated.

Is this far away from a release, just wanting an update? Planning to migrate my application to null safety but currently blocked by this at the moment.

okocsis, 2021-06-03:
> migrate my application to null safety but currently blocked by this at the moment.
@Jakegillingham5 
You can still migrate your code to a so-called "unsound null safety". I mean I'm using a null safe version of FlutterBleLib maintaining it on my own fork  https://github.com/privatdoktor/FlutterBleLib/tree/privatdoktor and even still my null safe migration isn't sound since there are still other dependencies not conforming to null safety. Perhaps time will fix these too.
I've also changed a couple of other features that needed fixing and enhancements, mostly on iOS side. I've more or less given up on waiting for Polidea for mering in my work. I might even rerelease the library to the public when it reaches a stable state.

Jakegillingham5, 2021-06-04:
@okocsis 
Awesome thanks so much for this, I actually made the call last week to just use your commit ['1beac74'](https://github.com/Polidea/FlutterBleLib/commit/1beac74c7fba5d1b0459f621bb98f26dca50cc59) and bring my project up to date ['unsound null safety' - Flutter 2.2.1]. Will move to your fork now though as most of our users are on iOS so will benefit from any changes in that space, cheers mate.

ifredom, 2021-08-03:
[image]

ozzy1873, 2021-09-02:
@okocsis 
Hi, I am using your fork too. A big thank you for making a null-safe version! By chance, have you tested it on Android 12? This still seems to be the best Flutter bluetooth plugin available. It makes me nervous that my project relies on a plugin that is no longer maintained, so it would be nice to know if you intend to maintain your fork or if you have suggestions for a different plugin.

okocsis, 2021-09-30:
> @okocsis Hi, I am using your fork too. A big thank you for making a null-safe version! By chance, have you tested it on Android 12? This still seems to be the best Flutter Bluetooth plugin [...]

@ozzy1873 I'm happy to hear that my branch works for others too, which branch of mine are you using in particular? 
In the Privatdoktor fork I'm currently working on the branch named new_native_ios_plugin. This branch is the one we use, but currently only supports iOS and since I changed/broke the flutter side API the android side needs the same work now in order to work correctly which is in progress now.
I'm also thinking of publishing my work under a new name and flutter plugin, since it will no longer share any codebase with this original one.
It will be easily portable from FlutterBleLib tho since the changes in the front-facing flutter API will be minimal.

Looking forward to early bird testers on both iOS (now stable) and android (under development). And any feedback and PR-s will be welcome too.

Best,
Oliver

ozzy1873, 2021-09-30:
Hi Oliver, thanks for the update! I am using the Privatdoktor fork. If the flutter API has only minimal changes, I would be happy to help test once the Android side is ready. 
John

From: okocsis
Sent: Thursday, September 30, 2021 4:50 AM
To: dotintent/FlutterBleLib
Cc: ozzy1873; Mention
Subject: Re: [dotintent/FlutterBleLib] Null-Safety migration (#556)

@okocsis Hi, I am using your fork too. A big thank you for making a null-safe version! By chance, have you tested it on Android 12? This still seems to be the best Flutter Bluetooth plugin available. It makes me nervous that my project relies on a plugin that is no longer maintained, so it would be nice to know if you intend to maintain your fork or if you have suggestions for a different plugin.
@ozzy1873 I'm happy to hear that my branch works for others too, which branch of mine are you using in particular?
In the Privatdoktor fork I'm currently working on the branch named new_native_ios_plugin. This branch is the one we use, but currently only supports iOS and since I changed/broke the flutter side API the android side needs the same work now in order to work correctly which is in progress now.
I'm also thinking of publishing my work under a new name and flutter plugin, since it will no longer share any codebase with this original one.
It will be easily portable from FlutterBleLib tho since the changes in the front-facing flutter API will be minimal.
Looking forward to early bird testers on both iOS (now stable) and android (under development). And any feedback and PR-s will be welcome too.
Best,
Oliver

PR #586 (2021-04-13): Nullsafety
Hey, guys! I've made a lot of effort to port the dart codebase to a nullsafe environment. I would appreciate someone taking a look at it.

known issues:
-not all of the test are compiling, although I've tried to make it work I've even ported some of the mocks to the new generated mocking scheme. FIXME: I think many of the test should be retired, in favour of general nullsafety. Please help me on that.
-there was a bleadapter lifecycle fiasco, because the DevicesBloc destroyed the client on dispose(), making the internal adapter unreachable, after going to detail view. Although I hav [... cut]