dotintent/FlutterBleLib #558: isConnected returns true for not paired device

fedotxxl, 2020-12-12:
Hello. I've got the following workflow on Android:

1. Manually pair devices (first time)
2. Manually unpair devices
3. Find device / connect / check connection

```
await peripheral.connect();
bool connected = await peripheral.isConnected();
```

On this step Android displays connection UI with pin input and `await peripheral.isConnected()` returns true when devices are not paired yet. 

Is it correct behavior? How to check that devices are really paired?

mikolak, 2020-12-12:
Yes - connection and pairing are two different things. I'd like to add support for pairing and checking its status, but I don't know when might that happen.
Right now your best bet is to attempt a read or a write to a characteristic that requires authentication.

fedotxxl, 2020-12-12:
ok, Thank you.

closed 2020-12-12 by fedotxxl

fedotxxl, 2020-12-12:
> Right now your best bet is to attempt a read or a write to a characteristic that requires authentication.

Can you explain the expected behavior? Exception?