dotintent/FlutterBleLib #563: Leaking BleAdapter

JamesMcIntosh, 2020-12-16:
If you create multiple clients when `FlutterBleLibPlugin.createClient()` overwrites the existing `BleAdapter`.

This means you can never destroy the previous clients, I'm pretty sure that this could cause a memory leaks as there are listeners attached etc.

michael-ottink, 2021-01-22:
@JamesMcIntosh did you test if this happens with the memory profiler?

JamesMcIntosh, 2021-01-24:
Hi @michael-ottink, I don't think that I checked it with a profiler, I came to this suggestion from a visual inspection of the code and the reported behaviour around hot restarts.

JamesMcIntosh, 2021-03-20:
I profiled it and it is leaking as expected.

JamesMcIntosh, 2021-03-20:
To get around the hot restart issue the createClient method can either destroy and recreate or reuse the existing one.

After testing out the destroyClient method and it looks like the destroy it still leaks.

The leak originates in MultiPlatformBleAdapter is because of RxJava1's subscriptions to Observers holds strong references to the anonymous inner classes inside BleModule i.e onError and onNext.

More information: https://medium.com/@scanarch/how-to-leak-memory-with-subscriptions-in-rxjava-ae0ef01ad361