dotintent/FlutterBleLib #565: Android Connection Status 133 Gatt Error - Cannot Connect

KrystofM, 2020-12-21:
Having trouble on some older Android devices with connecting to peripheral. For example Huawei P8Lite Android 6. The connection throws Gatt Error 133:
```
BleError (Error code: 201, ATT error code: null, iOS error code: null, Android error code: 133, reason: Disconnected from 8C:AA:B5:A7:62:4A with status 133 (GATT_ERROR), internal message: null, device ID: 8C:AA:B5:A7:62:4A, service UUID: null, characteristic UUID: null, descriptor UUID: null)
```
``Disconnected from 8C:AA:B5:A7:62:4A with status 133 (GATT_ERROR)``

This seems to be a problem in the function `connectToDevice` in `DeviceConnectingDelegate`.

Another flutter ble library has seem to solve this issue somehow: 
https://github.com/pauldemarco/flutter_blue/issues/231 
https://github.com/pauldemarco/flutter_blue/pull/229/files

Any ideas on how to solve this issue?

KrystofM, 2020-12-21:
This seems quite relevant also: https://github.com/Polidea/RxAndroidBle/issues/530

KrystofM, 2020-12-22:
The example in this repo simulates the behavior as well. This seems like an issue that can be solved either by trying to reconnect on fail or on Android 6 or less passing a an argument to gatt connection: https://stackoverflow.com/questions/25330938/android-bluetoothgatt-status-133-register-callback

KrystofM, 2020-12-22:
Okay to is seems RxAndroid lib is doing that when connecting without reflection: 
```
private BluetoothGatt connectGattCompat(BluetoothGattCallback bluetoothGattCallback, BluetoothDevice device, boolean autoConnect) {
        RxBleLog.v("Connecting without reflection");

        if (Build.VERSION.SDK_INT >= 23 /* Build.VERSION_CODES.M */) {
            return device.connectGatt(context, autoConnect, bluetoothGattCallback, TRANSPORT_LE);
        } else {
            return device.connectGatt(context, autoConnect, bluetoothGattCallback);
        }
    }
```

JAICHANGPARK, 2021-01-21:
I had same problem about this issues 
@KrystofM  did you solved this problem ?

Mushroomizer, 2021-02-22:
@JAICHANGPARK , @KrystofM 
I have the same issue, did you ever solve this?

KrystofM, 2021-02-26:
@JAICHANGPARK @Mushroomizer We have whitelisted a bunch of Huawei devices, it is a known problem for these devices to be problematic and even big companies like Garmin or Fitbit have them on a whitelist.

KrystofM, 2021-02-26:
Ovearll Status 133 Gatt Error is a pretty generic BLE error and could mean a lot of things. Had some similar problems with some Samsung devices and we had to change the firmware BLE stack on our device to be able to connect to it ( some really mysterious errors and issues come up with androids overall ).

matt39, 2022-01-21:
Did somebody solve this bug?

QGB, 2023-05-27:
connection failed gatt status 133

ManBali, 2023-07-17:
it impossible to solve this problems ,the only way to improve connect success, jus retry ,retry retry .

arbaelbarca, 2023-09-26:
please helping me, theres issue the same gat status 133

artisdom, 2024-07-23:
The solution posted above by KrystofM works fine for me,
basically you need to specify an additional parameter **"TRANSPORT_LE"** to **"connectGatt()"** Android call.

https://github.com/dotintent/FlutterBleLib/issues/565#issuecomment-749560900

```
        if (Build.VERSION.SDK_INT >= 23 /* Build.VERSION_CODES.M */) {
            return device.connectGatt(context, autoConnect, bluetoothGattCallback, TRANSPORT_LE);
        } else {
            return device.connectGatt(context, autoConnect, bluetoothGattCallback);
        }
```