dotintent/FlutterBleLib #568: startPeripheralScan doesn't return any results on iOS if trying to scan for a lightbar after disconnect.

n30phyte, 2020-12-30:
Hi!

I'm trying to implement an nRF DFU upgrade feature inside an app, and similar to #535, I try searching and it stops on it's own, not returning any results.

Currently the process is as follows:

1. Send device DFU command
2. Disconnect from device
3. Search for device in DFU mode via name
4. Get id (MAC/hash)
5. Send that identifier to another library.

On Android, 1-4 works fine. On iOS, this can only work if device is already in DFU and disconnected (done via another device/hardware switch).

I tested version 3.0.0-beta, 2.3.0, 2.2.x and it doesn't work on any of them.

n30phyte, 2020-12-30:
Solved the issue by adding a 750ms delay after disconnecting before searching for the device

closed 2020-12-30 by n30phyte

JasonEdinburgh, 2020-12-30:
This is on my todo list. Would you mind sharing the code? or a little more information?  

What library do you use in step 5?