dotintent/FlutterBleLib #572: could not find callback wrapper on Android

azakost, 2021-02-05:
Error occur when calling **startPeripheralScan()** function.

`D/BluetoothLeScanner( 5266): could not find callback wrapper`
`E/flutter ( 5266): [ERROR:flutter/lib/ui/ui_dart_state.cc(177)] Unhandled Exception: BleError (Error code: 600, ATT error code: null, iOS error code: null, Android error code: null, reason: Scan failed because application registration failed (code 6), internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)`

Same thing with an example app. Sometimes it works, but mostly throwing this kind of error.

Any ideas how to fix this?

ChrisMarxDev, 2021-03-24:
I am currently struggling with the same error on Android.

Some fixes that I encountered while searching for a solution were:

- Make sure that the BleManager is setup correctly 
`BleManager bleManager = BleManager();
await bleManager.createClient();`
- Make sure that all relevant bluetooth and location permissions are declared in the AndroidManifest.xml 
`    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>`
    
I am still experiencing this error, but maybe this helps you.

mikolak, 2021-04-07:
Have you requested the necessary runtime permission? From Android 10 you need ACCESS_FINE_LOCATION.