dotintent/FlutterBleLib #575: Flutter App scan not showing correct devices

patolax, 2021-02-28:
I just downloaded the repo and ran the example. The image below shows Bluetooth devices found on my PC, Android, IPhone and example flutter app running on android.

As you can see flutter app is not listing LAP-00 or my IPhone. But Android device scan is not showing my Iphone not sure why that is. 

Am I doing something wrong, why isnt the flutter app on android showing correct devices with the names? Is there a way to resolve this?

[image]

mikolak, 2021-02-28:
Hi!

IIRC the example app use local name from scan result instead of peripheral name. Try modifying code in devices_bloc, method _startScan(). Also make sure that those devices are BLE. Unless your iPhone is running a GATT server and is discoverable, it won't be found by the lib.

mikolak, 2021-04-07:
Closing as there're no follow ups.

closed 2021-04-07 by mikolak