dotintent/FlutterBleLib #582: Could not resolve Polidea:MultiPlatformBleAdapter

philos3, 2021-03-16:
it got wrong after i upgrade flutter to 2.0.1

AILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:processDebugResources'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve com.github.Polidea:MultiPlatformBleAdapter:0.1.8.
     Required by:
         project :app > project :flutter_ble_lib
      > Could not resolve com.github.Polidea:MultiPlatformBleAdapter:0.1.8.
         > Could not get resource 'https://jcenter.bintray.com/com/github/Polidea/MultiPlatformBleAdapter/0.1.8/MultiPlatformBleAdapter-0.1.8.pom'.
            > Could not HEAD 'https://jcenter.bintray.com/com/github/Polidea/MultiPlatformBleAdapter/0.1.8/MultiPlatformBleAdapter-0.1.8.pom'.
               > Read timed out

$ flutter doctor
Doctor summary (to see all details, run flutter doctor -v):
[√] Flutter (Channel stable, 2.0.1, on Microsoft Windows [Version 10.0.18363.1440], locale zh-CN)
[√] Android toolchain - develop for Android devices (Android SDK version 30.0.3)

 flutter_ble_lib: ^2.3.0

philos3, 2021-03-16:
flutter_ble_lib: ^2.3.2 is the same

closed 2021-03-16 by philos3

albo1337, 2021-03-25:
I'm facing the same problem. Could you tell me what you changed?

albo1337, 2021-03-25:
I've fixed the problem. How could you close the ticket without posting your solution? Anyways, what I've done:

- update version of flutter_ble_lib in pubspec.yaml
- flutter clean
- remove podfile.lock
- cd ios
- pod repo update

Then everything worked again and I was able to successfuly build again.