dotintent/FlutterBleLib #591: Descriptor operations return only partial Characteristic

mikolak, 2021-04-13:
Methods returning Descriptors (`descriptorsForPeripheral/Service`, `readDescriptorForPeripheral/Service`) send only partial information about characteristic, ie. they're missing characteristic's properties, which results in errors due to type safety (null is not bool/String).

okocsis, 2021-04-14:
> Methods returning Descriptors (`descriptorsForPeripheral/Service`, `readDescriptorForPeripheral/Service`) send only partial information about characteristic, ie. they're missing characteristic's properties, which results in errors due to type safety (null is not bool/String).

So it seems, thar null safety did reveal some of the hidden uncovered cases, causing bugs, already! 
that’s a good thing!
hope we can fix these issues quickly

DevNico, 2021-04-27:
Any timeline on when you could get this done? Would be really great to have a nullsafe release