dotintent/FlutterBleLib #592: Passing null to BleManager.cancelTransaction() on iOS hard-crashes the entire app without a useful stack trace

Lootwig, 2021-04-16:
Should maybe be checked for before passing the parameter to any native API.

okocsis, 2021-04-22:
nullsafety will make this impossible in 3.0 release

okocsis, 2021-05-12:
> @okocsis hi, is there any tentative date for 3.0 nullsafety release? This is a great library :)

I’m no collaborator here, I just did a PR from my own fork.
I don’t know but you can always use the latest commit on dev branch, or even fork that, and use the lib referring to a local git submodule in your pubspec.yaml.

On privatdoktor’s fork I’m working on a whole new ios layer, it will be a simple and elegant implementation using purely CBCentralManager and CoreBluetooth framework and Swift.

I had issues with the current ios implementation various race conditions and strange behaviours, and i had to realise it’s way too over complicated to easily patch these issues, meaning it’s nearly impossible to debug if there is 4layers of pub and pod libraries in between the app and the actual CoreBluetooth framework.

It will now just be 2layers of me own shit not someone else’s :P /just kidding/
so guys feel free to check out my work on my fork.
cheers

fpabl0, 2021-05-12:
@okocsis Sounds great, thanks for sharing it :), I will take a look. Are you planning to merge your work (when it is finished) in this repository or just keeping it isolated?

okocsis, 2021-05-12:
> @okocsis Sounds great, thanks for sharing it :), I will take a look. Are you planning to merge your work (when it is finished) in this repository or just keeping it isolated?

will surely make a PR when it’s ready, but now first priority is for our app to make it function on ios properly.
thus i cannot guarantee a comprehensive testing scheme although i am doing my best here.
I would really appreciate others testing it and reporting issues too!

thanks in advance!