dotintent/FlutterBleLib #593: peripheral.discoverAllServicesAndCharacteristics() finds cached (outdated) services

Simpler1, 2021-04-24:
I'm building a BLE server and as I add or remove services, they are not found through the `peripheral.discoverAllServicesAndCharacteristics()` and `peripheral.services()` commands.

Other Bluetooth scanners such as _nRF Connect_ and _LightBlue_ both find the changed (added/removed) services, but _FlutterBleLib_ does not.

I've tried this with both 2.3.2 and 3.0.0-beta and both have the same issue.
I've tried flutter clean.
Uninstalled the flutter app and reinstalled through debug.
I even tried the example app from FlutterBleLib, but it also shows the old services (some of which don't even exist on the server any more).

Is there some other way to clear the cache and have it actually read the services again?

Simpler1, 2021-04-26:
Using the big hammer method of resetting the factor defaults on the phone allowed it to find the updated services, but there must be a simpler way.

mikolak, 2021-04-26:
Android or iOS?

jobfeikens, 2021-04-26:
`peripheral.connect(refreshGatt: true)`

Simpler1, 2021-04-26:
Thanks guys...
Platform is Android.  I'll try the `refreshGatt: true`.

Simpler1, 2021-04-27:
Problem resolved.

closed 2021-04-27 by Simpler1