dotintent/FlutterBleLib #598: Request support the null-safety

JonYun, 2021-05-29:
Dear FlutterBleLib team, I am very eager for this library to support null-safety

ifredom, 2021-08-09:
+1  . wait for null-safety

leech001, 2021-08-29:
Add on pubspec.yaml
```
  flutter_ble_lib:
    git:
      url: git://github.com/Polidea/FlutterBleLib
      ref: develop
```

ifredom, 2021-09-03:
> Add on pubspec.yaml
> 
> ```
>   flutter_ble_lib:
>     git:
>       url: git://github.com/Polidea/FlutterBleLib
>       ref: develop
> ```

My search found that there is no branch of the library fluterblelib in polidea. Was the address sent wrong?

MoralCode, 2021-11-18:
it appears the current dev branch supports it courtesy of #586, just hasn't been released to pub.dev yet

ifredom, 2021-11-25:
> it appears the current dev branch supports it courtesy of #586, just hasn't been released to pub.dev yet

Thank you for your reply. I'll try