dotintent/FlutterBleLib #602: nRF52 UART can read, but can't enable notifications or write descriptor

pavel604, 2021-06-22:
I have nRF52832 programmed with OEM UART code example. 
I have no problems connecting and writing to peripheral, but I can't enable notifications, read RX characteristic or write to descriptor.

When I call `bleRxCharacteristic.monitor(transactionId: "1")` I get positive response: `setCharacteristicNotification() - uuid: 6e400003-b5a3-f393-e0a9-e50e24dcca9e enable: true`, but if I read descriptor 0x2902 - I get `[0,0]`, which means notifications are still disabled. So the `monitor` method didn't work.

If I try to write to descriptor to enable notification, like `await blePeripheral.writeDescriptor(bleServiceId, bleRxCharacteristicId,
        bleNotifDescriptorId, Uint8List.fromList([enable ? 1 : 0, 0]));`, - I get BleError 506.

I hope that makes sense.
Also documentation is fuzzy on bonding, so I can't try to bond and read.

Hardware: Google Pixel 5 with Android 11.

nRF-specific data:
`final String bleServiceId = '6E400001-B5A3-F393-E0A9-E50E24DCCA9E';
  final String bleTxCharacteristicId = '6E400002-B5A3-F393-E0A9-E50E24DCCA9E';
  final String bleRxCharacteristicId = '6E400003-B5A3-F393-E0A9-E50E24DCCA9E';
  final String bleNotifDescriptorId = '00002902-0000-1000-8000-00805f9b34fb'; //this is where you set [1,0] to get notifications`

Thanks.

closed 2021-06-24 by pavel604

mtizim, 2021-09-20:
@pavel604 I am having a similar issue. Could I ask how you fixed your problem?

pavel604, 2021-09-23:
Firmware needed tweaking. I don't know exactly what the embedded dev did, unfortunately. but it was not the ble plugin that caused the headache