dotintent/FlutterBleLib #609: Unhandled Exception: BleError (Error code: 2, ATT error code: null, iOS error code: null)

sivaperumal644, 2021-09-02:
While trying to connect device it throws the below error sometimes and not always. I couldn't identify what causes this.
 
`[ERROR:flutter/lib/ui/ui_dart_state.cc(199)] Unhandled Exception: BleError (Error code: 2, ATT error code: null, iOS error code: null, Android error code: null, reason: null, internal message: null, device ID: null, service UUID: null, characteristic UUID: null, descriptor UUID: null)`

The error code does not remain the same it changes to 2, 205, 204 and many but couldn't handle this error.

I tried `connect(requestMtu: 250).catchError((){print("some error")})` and `connect(requestMtu: 250).onError((error, stacktrace) {print("error");});`. But it didn't go into catch or the error block.

Couldn't identify what is causing this exception to occur at the first place, and also don't how to handle this exception.

ride4sun, 2021-12-21:
Get the same error - I think it is known by the package owner but he is not fixing it. I am wondering if someone branched the package and fixed all the issues because otherwise, this package is solid.

skashok, 2022-06-03:
Hi @sivaperumal644  @ride4sun 

Did you all manage to get workaround for this issue. I am too still getting this error.

freedreamer82, 2022-06-03:
same here...please find a workaround fix :(

MoralCode, 2022-06-10:
there was some discussion of alternatives in #613. although i'm not aware of anyone who has fixed this particular issue.

On the bright side, this seems to be a list of what the codes mean: https://github.com/dotintent/FlutterBleLib/blob/a4df42eda471f0c53a06b7b9937db8b59585f318/lib/error/ble_error.dart#L55

TL;DR: 
2 -> operationCancelled
204 -> deviceNotFound
205 -> deviceNotConnected

that might help solve your more immediate issue

sivaperumal644, 2022-06-11:
> Hi @sivaperumal644 @ride4sun
> 
> Did you all manage to get workaround for this issue. I am too still getting this error.

I couldn't find a solution and I had few more deprecated issues in both android and iOS with this plugin, So I decided to use a different plugin. I used [flutter_blue_plugin](https://pub.dev/packages/flutter_blue_plus).