dotintent/FlutterBleLib #616: Build error with Xcode 13,  iOS 15 SDK

wxsdmy, 2021-11-04:
…/RestoredState.swift:68:105: error: value of optional type 'CBPeripheral?' must be unwrapped to a value of type 'CBPeripheral'
                                                      peripheral: RxCBPeripheral(peripheral: $0.service.peripheral)),
                                                                                                        ^
…/MultiplatformBleAdapter/iOS/RxBluetoothKit/RxCBCharacteristic.swift:59:52: error: value of optional type 'CBService?' must be unwrapped to a value of type 'CBService'
        return RxCBService(service: characteristic.service)
                                                   ^                                                      peripheral: RxCBPeripheral(peripheral: $0.service.peripheral)),

…/RxCBDescriptor.swift:43:62: error: value of optional type 'CBCharacteristic?' must be unwrapped to a value of type 'CBCharacteristic'
        return RxCBCharacteristic(characteristic: descriptor.characteristic)
                                                             ^

Please update MultiPlatformBleAdapter to '0.1.9' version.

freedreamer82, 2021-11-17:
any news ? we should need a version compatible with ios 13!

b055man, 2021-11-22:
Try this: https://github.com/dotintent/MultiPlatformBleAdapter/issues/79#issuecomment-945158358

MoralCode, 2021-12-22:
it appears that there is a fork of this project explicitly for supporting iOS 15 at https://github.com/davejlin/flutter_ble_lib_ios_15