dotintent/FlutterBleLib #618: please update to null safety, thank you!

GarrettLi77, 2022-01-04:


matteocordray, 2022-04-01:
What's the work around for this?

brian-quah, 2022-05-05:
@matteocordray 
> What's the work around for this?

As with any project that uses plugins that have yet to be migrated to null safety, run/build with the option  `--no-sound-null-safety`

For example:
To build apk: `flutter build apk --no-sound-null-safety`
To run: `flutter run --no-sound-null-safety`
To integrate with VSCode IDE launching:
In `.vscode/launch.json`, Add `"--no-sound-null-safety"` under the `configurations.args` array.

matteocordray, 2022-05-05:
@brian-quah Thank you! I ended up just moving to another library instead. I didn’t want to move away from null safety.

ziyadmansy, 2022-07-04:
> @brian-quah Thank you! I ended up just moving to another library instead. I didn’t want to move away from null safety.

Would you tell me which package you have used instead?

matteocordray, 2022-07-04:
It's called Flutter Reactive BLE. It's apparently made by Phillips Hue so it seems like it will have decent support:

[Flutter Reactive BLE Library](https://pub.dev/packages/flutter_reactive_ble)