dotintent/FlutterBleLib #622: Android 12 error: could not find callback wrapper

lpc-nz, 2022-05-17:
Hi everyone
I am working on the Bluetooth connection and I have some problems with the new Android version.
Most Android phones work perfectly except for some Samsung phones (S21, S21 +).
Does anyone know about these issues? Please give me some help.

```
I/flutter (23844): DeviceService - waiting for power on - BluetoothState.POWERED_ON
I/flutter (23844): DevicesScanningBloc - start scan
D/com.polidea.flutter_ble_lib.FlutterBleLibPlugin(23844): on native side observed method: startDeviceScan
I/BluetoothAdapter(23844): STATE_ON
I/BluetoothAdapter(23844): STATE_ON
I/BluetoothAdapter(23844): STATE_ON
I/BluetoothAdapter(23844): STATE_ON
D/BluetoothLeScanner(23844): Start Scan with callback
I/BluetoothAdapter(23844): STATE_ON
I/BluetoothAdapter(23844): STATE_ON
D/BluetoothLeScanner(23844): could not find callback wrapper
```

iJack93, 2022-11-09:
Any update on this?

georgechem, 2023-03-31:
Have the same issue with Bluetooth on android 12, on 10 and 11 works perfectly. Adding additional permissions required by Android 12 and updating OS to the latest version did not change anything.

iJack93, 2023-03-31:
@georgechem I've found that you have to add at first these permissions inside the AndroidManifest and through a Permission manager ask for them to the user

```
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
``` 

Then you have to target at least api 33 of Android, otherwise it won't work and won't event ask for those.
I think that this is not related to this library, but to Android itself.

georgechem, 2023-03-31:
[image]

It did not work - looks like permissions are granted but nothing happen beside this. It is SONY-XQ-AS-52

rohailmansab, 2023-10-07:
This solution is not working

rohailmansab, 2023-10-07:
[image]
This is my Flutter app error in the Bluetooth. Can someone solve my error?