dotintent/FlutterBleLib #625: iOS gets Error code: 402 when reading data from monitorCharacteristic()

Aldaniee, 2022-11-08:
[VERBOSE-2:dart_vm_initializer.cc(41)] Unhandled Exception: BleError (Error code: 402, ATT error code: 2, iOS error code: null, Android error code: null, reason: Reading is not permitted., internal message: null, device ID: 48E32FDC-BC25-0B46-AC03-549FD00B26B8, service UUID: ce060030-43e5-11e4-916c-0800200c9a66, characteristic UUID: ce060039-43e5-11e4-916c-0800200c9a66, descriptor UUID: null)

Android works with no errors but iOS throws this when data is received in the listen() from monitorCharacteristic steam.

MoralCode, 2022-11-10:
looks like this already has an answer in #549 
> The characteristic you're trying to read does not support reading, only notifications. Use .monitor() and listen to the stream.

MoralCode, 2022-11-11:
unable to reproduce on iOS 15.0.1 or 15.7.1. seems like it may be limited to iOS 16 in some way.
it seems like there is also precedent for iOS doing weird stuff from version to version given that a fork (https://github.com/davejlin/flutter_ble_lib_ios_15) seems to have been created by someone who needed iOS 15 support

Aldaniee, 2022-11-11:
I have now tested this with an iPhone 13 Pro on 16.0.0 and an iPhone 12 Pro on 16.0.2 and found the same issue on both further enforcing the idea this is an iOS 16 specific issue. I'll change the title to reflect this.